NVIDIAGameWorks/PhysX #655: Crash in IslandSim

zemlifr, 2024-01-04:
Hello, we have encountered crash in IslandSim code with following callstacks

```
>	PhysX_64.dll!physx::IG::IslandSim::activateIsland(unsigned int islandId) Line 649	C++
 	PhysX_64.dll!physx::IG::IslandSim::processNewEdges() Line 1052	C++
 	PhysX_64.dll!physx::IG::SimpleIslandManager::firstPassIslandGen() Line 206	C++
 	PhysX_64.dll!physx::Cm::Task::run() Line 67	C++
```
And

```
  PhysX_64.dll!physx::IG::IslandSim::addConnectionToGraph(unsigned int handle=9) Line 273        C++
         [Inline Frame] PhysX_64.dll!physx::IG::IslandSim::insertNewEdges() Line 890        C++
         PhysX_64.dll!physx::IG::IslandSim::processNewEdges() Line 927        C++
         PhysX_64.dll!physx::IG::SimpleIslandManager::firstPassIslandGen() Line 206        C++
         PhysX_64.dll!physx::Cm::Task::run() Line 67  
```

Unfortunately it happens very rarely in quite complex multiplayer scenario, so we don't have any simple or stable repro.
I only suspect from some other logs that it happens when some joint constraint is violated which results to huge force applied to connected object.

I can see velocities like (-33018.468750, 107253.906250, 26213.623047) and it seems CTD happens several frames later, not immediately.

Is there some fix for that, or at least some tip which might cause that so we can try investigate deeper?

Thanks

preist-nvidia, 2024-01-08:
Hi Frantisek,

Can you repro it in PhysX 5, assuming you are using 4? But sounds like it's not easy to use PhysX 5 if it only happens in the complex multiplayer scenario.

Thanks,
Philipp

zemlifr, 2024-01-09:
Hi, thanks for reply. Yes we are using 4.1.2. I have some prototype where we use Physx 5, but it is very wip. 
Crash really happens rarely during release testing, so there is not much chance to try physx 5 right now or even add some heavy logs :(