NVIDIAGameWorks/PhysX #656: Strange URLs on NVIDIA PhysX Website

lordbendtner1337, 2024-01-18:
There are some really strange URLs on the NVIDIA PhysX website; https://www.nvidia.com/en-us/drivers/physx-downloads/

The "PhysX Developers Website" button redirects you to https://supportcenteronline.com/ics/support/default.asp?deptID=1949

The "Tool Vendor Verification System" button redirects you to https://www.warioworld.com/tvvs/nvidia/

preist-nvidia, 2024-01-22:
Hi @lordbendtner1337 - thanks for the report, I see the weird supportcenter link, but cannot access. Will investigate!

preist-nvidia, 2024-01-24:
Thanks again for the report, we'll replace this obsolete page with a redirect to our current SDK landing: https://developer.nvidia.com/physx-sdk

closed 2024-01-24 by preist-nvidia