NVIDIAGameWorks/PhysX #657: Crash in releasing Scene Queries

CuSO4H2O5, 2024-01-31:
Hello, I have encountered a weird crash on using PhysX 3.4.1. But I don't know how this triggered.
When I switch the phyx scene between my 2 levels, which contains a pxscene release and a pxscene create, it sometimes crash because of a simd overflow in releasing AABB tree of scene query.  mNodeAllocator, mParentIndices pointer, mTotalNbNodes, mTotalPrims, this four in AABBTree are set to 1.0 in floating point, so it might be a simd overflow.
I have tried simulate once before release to update its sq structure, but it not works. And I found that it will always crash when AABB Trees are empty. 
Unfortunately it happens just in a complex editor, so I can't make a simple repo of it. I want to know if physx team has already fixed this issue or just my miss used it?
The crash stacks are as follow picture. Could someone helps? thanks a lot.
[image]

preist-nvidia, 2024-01-31:
Hi Copper Sulfate, I assume you are not easily able to switch to our newer PhysX 5?

I'm adding @PierreTerdiman who might have a deeper insight about the crash. Internal tracking: PX-4782.

PierreTerdiman, 2024-02-07:
I'm afraid I don't have a deeper insight into this one. PhysX 3.4.1 was a long time ago now, and while I don't immediately recognize this specific crash / callstack, it is possible that it has been fixed in newer versions. Would it be an option to upgrade to a more recent PhysX build ? If not, are you able to recompile PhysX 3.4.1 anyway ? It could be an issue in that "extended bucket pruner" and IIRC there was a way to disable it with a compile time flag or define.

preist-nvidia, 2024-04-23:
Closing for now.

closed 2024-04-23 by preist-nvidia