NVIDIAGameWorks/PhysX #659: Inverse Mass Scale Contact Modification and CCD

PhilMaguire, 2024-04-10:
Hi there,

I am using PhysX 3.4 with Unreal Engine 4.27.

I am successfully modifying contacts on bodies with CCD enabled. I am having an issue, however, modifying the contact for a particular collision.

I want a particular street sign to behave as if it is very light (1 kg) when a vehicle hits it but otherwise have a realistic sort of mass. I thought I could achieve this with an inverse mass scale and inverse inertia scale during contact modification. Despite modifying these values, however, I still lose a lot of speed (maybe 13mph) when hitting the sign at 100mph with my ~2000kg vehicle.

If I disable CCD then the contact modification of the inverse mass and inertia scale behaves as I would expect and I don't noticeably lose any speed when hitting the sign. 

I can't disable CCD to fix the issue as this causes all sorts of other unwanted problems.

Here is some output capturing the problem

[2024.04.10-09.13.11:386][146]LogTemp: LogVehicleForce Post Vehicle Update vel V(X=685.11, Y=**-4759.98**, Z=21.15) (4809.075) angular V(X=0.00, Y=0.00, Z=0.00) (0.001)
[2024.04.10-09.13.11:387][146]LogTemp: LogVehicleForce modifying contact 0 between Road_Sign_USA_Speed_C_2:StaticMesh and BP_Vehicle_Muscle_1_C_0:AdditionalCollisionVersusVehicles
[2024.04.10-09.13.11:387][146]LogVehicleForce: Verbose: [1147] PLAYER VEHICLE FORCE - CONTACT MODIFICATION - InvMassScale and InvInertiaScale 0.004 applied to BP_Vehicle_Muscle_1_C_0 by ADangerZoneWheeledVehicle::OnContactModify. Vel V(X=685.11, Y=**-4759.98**, Z=21.15)
[2024.04.10-09.13.11:388][146]LogTemp: LogVehicleForce OnContact WITH Notify between Road_Sign_USA_Speed_C_2:StaticMesh (Sign_Small_Retangular) Vel V(0) and BP_Vehicle_Muscle_1_C_0:AdditionalCollisionVersusVehicles (Musclecar_DD2_CollisionCar2Car) Vel V(X=685.11, Y=**-4759.98**, Z=21.15).
[2024.04.10-09.13.11:389][146]LogTemp: LogVehicleForce ATFEWheeledVehicle::NotifyHit Start Vel V(X=595.26, Y=**-4153.58**, Z=0.55)

- After the PhysXVehicle manager has updated the vehicle my speed in the y-axis is -4759.98 cm/s
- I then modify the contact applying an inverse inertia and mass scale of 0.004 to the vehicle rigid body, at this point the speed in the y-axis is still the same
- I then log the contact in a PxSimulationEventCallback called onContact. At this point the speed in the y-axis is still the same
- At the point this contact is reported to my game code (via the UE4 NotifyHit function) the speed has reduced in the y-axis significantly despite there being no other contacts and no vehicle updates run

There is no (noticible) speed loss when the same code is run with CCD disabled on the vehicle rigid body.

In a PVD capture my road sign body looks like this:

[image]

In a PVD capture my vehicle body looks like this:

[image]

The "boxier" highlighted shape is what is configured to collide with the road sign rigid body:

[image]

Any help would be greatly appreciated.

Kind regards,

Phil

vreutskyy, 2024-04-16:
Hi Phil
Just a quick question, did you try using dominance groups?
https://docs.nvidia.com/gameworks/content/gameworkslibrary/physx/guide/Manual/RigidBodyDynamics.html#dominance

It seems like an alternative solution to the contact modification.

PhilMaguire, 2024-04-16:
I wasn't aware of dominance groups. Thank you for pointing them out as they sound like a good alternative for this case.

PhilMaguire, 2024-04-16:
I have tried out dominance groups and they work well for stopping my vehicle losing speed when hitting the road sign.

I tried a very simplistic test of putting my vehicle in group 1 and then having group 1 dominate group 0 (the default group) i.e.

`mScene->setDominanceGroupPair(1, 0, PxDominanceGroupPair(0.f, 1.f));`

This had the desired effect on the sign but also had some rather undesirable effects such as my vehicle falling though the landscape. I guess this is what would be expected given the vehicle now dominates everything else. Presumably the way to avoid this is to add a group 2 and specifically put things I don't want to affect my vehicle velocity into this group and then have group 1 dominate group 2, rather than group 0?

Thanks,

Phil

vreutskyy, 2024-04-17:
Yes, sure this dominance group hierarchy should be thought through to avoid undesired effects.
Depending on the content of your game world, you could define several dominance levels like: 0 - world, 1 - vehicle, 2 - signs, and set that 0 dominates 1 and 1 dominates 2. Then group 0 would have all your static and kinematic world shapes, group 1 - all vehicles and big obstacles that can push and be pushed by a vehicle, and group 2 is for small objects like signs or debris that can be pushed by the vehicle, but cannot push back.
Of course, you can add even more dominance levels. Up to 32.
Hope this helps.

PhilMaguire, 2024-04-17:
Thank you that is really helpful.

closed 2024-05-22 by vreutskyy