NVIDIAGameWorks/PhysX #660: Enabling eMODIFY_CONTACTS causes jitter on wall collision.

kujukuju, 2024-06-20:
I'm trying to manually manipulate some contacts in my game, so I've set up a contact modify callback and enabled the eMODIFY_CONTACTS scene flag. I'm using the rust version of physx which is 4.1.

For reference, I am able to modify contacts and things work as expected.

However, when I enable modify contacts all collisions in my scene where I push up against a wall begin to jitter. This happens even when I do completely nothing in my PxContactModifyCallback.

I couldn't find another reported issue on this github about this. From what I can tell, it's because modifying contacts causes the physics engine to generate new patches.

It seems like this is an existing issue, because I found this user on a forum who reported my exact issue.
https://gamedev.net/forums/topic/715265-physx-rigidbody-jitters-if-pushed-into-corners-depenetration-issue/

He ended up just disabling contact modification and stopped there. So unfortunately not helpful for me.

I actually went into the source code and commented out the area where it seems to force a patch update when contacts are modified and this did nothing. I wanted to go through all the source code step by step and find exactly where the bug was being caused but I actually couldn't find anywhere eMODIFY_CONTACTS was used to perform additional steps that seemed to affect the jitter bug at all.

However, I can confirm as shown in my video, as with the gamedev.dev guy, the bug only happens when the contact modification flag is enabled. Nothing else matters at all.

Here is a video showing the bug. I have eMODIFY_CONTACTS enabled at the start (with no actual contact modification happening at all), and once it's disabled the bug goes away.

https://github.com/NVIDIAGameWorks/PhysX/assets/5142257/860f6878-6f8f-42fd-a05a-d760b26dfa09

Any advice on how to proceed here would be very appreciated!

kujukuju, 2024-06-20:
I also want to add some clarification that the character you see here is just a basic dynamic rigid body capsule. No kinematics, no character controllers.

I just began the process of switching my physics over to physx, so my physics scene setup is currently incredibly simple.

I've also tested this with custom convex mesh geometry and experience the same issue with modify contacts on/off.

kujukuju, 2024-06-21:
The issue seems to be caused by the modifiable contacts having a non-expected restitution. I'll close this but leave up the issue on omniverse because it seems like a real bug.

closed 2024-06-21 by kujukuju

vreutskyy, 2024-06-24:
Thanks. Answered in omniverse's repo.