NVIDIAGameWorks/PhysX #661: Point in Polygon problem

ViaCognita, 2024-06-28:
Hi!

First of all, I don't know if this is the right place to ask this question because I have no idea about NVIDIA PhysX.

I'm developing a game with Unreal Engine 5.4.2. with C++ and I need to find if a point is inside a polygon. Searching on Internet I have found someone mentioning that it can be done with PhysX. On Wikipedia, I have found that the Ray Casting algorithm or the Winding number algorithm can do it. 

If I have a set of 2D points, how can I know if another point is inside the polygon defined by that set of points using PhysX? If it is possible.

Thanks!

vreutskyy, 2024-06-28:
Hi @ViaCognita 
No, this isn't the right place. PhysX is a physics simulation library, and what you need is some geometry query library or just a function. I think the easiest thing you can do is just ask ChatGPT to write the function for you. It should be pretty simple.
Cheers

PierreTerdiman, 2024-06-28:
https://erich.realtimerendering.com/ptinpoly/

RaghavDabra, 2026-10-28:
Appreciate the pointer to Eric Haines' repository! That link covers pretty much every algorithm needed for this. Cheers!