NVIDIAGameWorks/PhysX #667: Which parameters can I set to change the contact performance during the modify contact?

pgpgwhp, 2024-08-14:
### The Problem
Exactly. I want to change the influence of the specified contacts, thus, during the modify contact callback after the narrow phase , I use the setMaximpluse for each contact points. However, in some cases, the vel of the dynamic objects is to so fast, What should I do to make it more slower?
  
### My solution

1.  I do some test in the PhysX 4.1 and PhysX 3.4,  May the change of the restitution make influence to the vel of different contact bodies?
2.  Directly setLinearVelocity for the contact dynamic bodies can change the velocity during the modify contact, and it will  apply for the PGS in dynamicSolver. But the gravity and some other forces will make no influence to the velocity as I tested , it's right to use the setLinearVelocity during the modify contact ?

pgpgwhp, 2024-08-15:
### Another Question
I test the overlapping dynamics bodies and set the contact Maximpluse between them as zero. As we expected, the contact is ignored, However. when I printf the count number of the pairs in contact modify callback, When the dynamic bodies contact with the static plane, the  contact modify callback will happen two times in one simulation time. It's because the contact between the rigid bodies and the static plane is new?  I change some codes in the example of snippet of contactModify put my codes as below ：

`class ContactModifyCallback: public PxContactModifyCallback
{
	void onContactModify(PxContactModifyPair* const pairs, PxU32 count)
	{
#if MODIFY_MASS_PROPERTIES
		//We define a maximum mass ratio that we will accept in this test, which is a ratio of 2

		printf("Count %d \n", count);
		for(PxU32 i = 0; i < count; i++)
		{
			const PxRigidDynamic* dynamic0 = pairs[i].actor[0]->is<PxRigidDynamic>();
			const PxRigidDynamic* dynamic1 = pairs[i].actor[1]->is<PxRigidDynamic>();
			if(dynamic0 != NULL && dynamic1 != NULL)
			{
				printf("test PxRigidDynamic \n");
				for(int j = 0; j < int(pairs[i].contacts.size()); j++)
				{
					pairs[i].contacts.setMaxImpulse(j , 0.f);
				}
			}
		}
#endif
	}
};`

`PxRigidDynamic* createDynamic(const PxTransform& t, const PxGeometry& geometry, const PxVec3& velocity=PxVec3(0))
{
	PxRigidDynamic* dynamic = PxCreateDynamic(*gPhysics, t, geometry, *gMaterial, 10.0f);
	dynamic->setAngularDamping(0.5f);
	dynamic->setLinearVelocity(velocity);
	PxRigidBodyExt::updateMassAndInertia(*dynamic, 5.0f);
	gScene->addActor(*dynamic);
	return dynamic;
}`

`	// createStack(PxTransform(PxVec3(0,0.0f,10.0f)), 5, 2.0f);
	createDynamic(PxTransform(PxVec3(-60,21,4)), PxSphereGeometry(20), PxVec3(0,0,0));
	createDynamic(PxTransform(PxVec3(-60.f,21,6.f)), PxSphereGeometry(20), PxVec3(0,0,0));
	createDynamic(PxTransform(PxVec3(-60,31,3)), PxSphereGeometry(20), PxVec3(0,0,0));
	createDynamic(PxTransform(PxVec3(-60,31,3)), PxSphereGeometry(20), PxVec3(0,0,0));`

and it will print such reports
`Count 8 
test PxRigidDynamic
test PxRigidDynamic
test PxRigidDynamic
test PxRigidDynamic
test PxRigidDynamic
test PxRigidDynamic
2 contact reports
Count 8 
test PxRigidDynamic
test PxRigidDynamic
test PxRigidDynamic
test PxRigidDynamic
test PxRigidDynamic
test PxRigidDynamic
Count 2
4 contact reports`

vreutskyy, 2024-08-16:
Hi @pgpgwhp 
Sorry, I don't understand what you're trying to do. Could you maybe describe step by step what's happening and what should happen instead?