NVIDIAGameWorks/PhysX #670: setMaxJointVelocity on a joint makes the whole articulation rigid

ilyal451, 2024-10-12:
Hi! I'm trying to create a motor on an articulation that will stop dead when the target velocity is zero. However it's affected by the gravity and other forces and gives little by little (starts drifting) despite the motor force is big and the friction is set to a large value. So I'm trying to disable the joint completely by setting the setMaxJointVelocity to 0. It starts working as expected (almost, a bit jerky) but the whole articulation becomes kind of rigid (like if having a very high friction in every joint). Is it normal?

Also I'd like to know if there's any other way I can achieve the desired behavior.

Thanks.

ilyal451, 2024-10-15:
Setting the joint to track position worked, thanks! But I currently found a different solution: increasing the gains to some ridiculously high values worked for me..

Anyways, what's it for? Why setting the max joint velocity on one joint affects the whole articulation? Is this a feature or a bug (or I'm doing something the wrong way)?

> Hi @ilyal451 - if you try to keep a joint at a position via velocities, you likely get drift. I would recommend setting zero stiffness while in velocity control mode, and then enabling a position drive to hold the motor at a target position when the drive should stop.

preist-nvidia, 2024-11-22:
It's just not a good approach to control position using velocity becuase of the drift. Glad you figured out he workaround with the position drive which is how I would do it too.

closed 2024-11-22 by preist-nvidia