NVIDIAGameWorks/PhysX #672: Gear joint attached to an articulation tends to make it start creeping all over the place

ilyal451, 2024-10-15:
I attached a gear joint to several articulation revolute links (one is motored), all having an unlimited rotation around TWIST. After some time (of seemingly working fine) the whole thing starts to behave strange: like if some force dragging it here and there over the ground.. The hinges on the gear joint are set properly (I believe). I'm trying to model a car transmission: engine, transmission, wheels. So it is attached in this order (having several gear joints, correct).

I noticed the same thing while attaching a different joint (6DOF) to an articulation, which limits were (rigidly) violated (weren't set up properly). The problem gone when I fixed the limits, but I don't see what I can fix with the gear joint (it worked fine for some time, as I already mentioned).

vreutskyy, 2024-10-15:
Hi @ilyal451 
It would help if you could provide a minimal runnable snippet/sample so we could better understand the problem. Thanks.
Also, the latest PhysX 5 repository is here [https://github.com/NVIDIA-Omniverse/PhysX](https://github.com/NVIDIA-Omniverse/PhysX). PhysX 4 isn't updated anymore as fay as I know.

ilyal451, 2024-10-15:
Hi @vreutskyy I'll try to do it on this week (I need to see, maybe I can reuse PhysX samples/snippets for this). Yes, it's PhysX 5.4.2.