NVIDIAGameWorks/PhysX #674: Direction dependent friction

ilyal451, 2024-10-24:
Hi! How can I have a direction dependent friction in PhysX? I have a setup like this: a cylinder with grooves along its axis of rotation (like a gear) which while being put on a surface will have more friction while rotating around this axis than while sliding along it.

vreutskyy, 2024-10-24:
Hi @ilyal451 
It's usually called anisotropic friction, and unfortunately PhysX currently doesn't have it. You can try using contact reporting to apply some forces to achieve the desired behavior. Or if this cylinder isn't just some freely moving object but a part of a mechanism you can limit its motion with one of existing or make a custom joint. There can be different approaches depending on the final goal.

ilyal451, 2024-10-25:
@vreutskyy It worked, thanks a lot!

One more thing: I'm calculating the counter impulse as Basis * ( FrictionImpulse dot Basis ). Is that correct? It's just that I noticed some slowing down in the direction I didn't want to affect (which is though easy to mitigate by just applying more initial torque). The problem is that the more I scale (the friction) the more it drops, to the point that sometimes it stops moving in the desired direction completely (especially when actively slipping in the other direction). In all other respects it works as expected.

vreutskyy, 2024-10-31:
Hi @ilyal451, sorry for the delay
Yes, that was the idea, but it's hard to say what goes wrong without seeing what is actually going on :)

ilyal451, 2024-10-31:
@vreutskyy I'm going to make a demo soon where I'll put all the issues I've currently encountered (just haven't had enough time yet). Can you please keep me informed if you've already found and fixed anything, so that I don't do unnecessary job?

vreutskyy, 2024-10-31:
Sure.

But you really should create all this issues in the correct repository
https://github.com/NVIDIA-Omniverse/PhysX

People may think it's all about the old version of PhysX

ilyal451, 2024-10-31:
@vreutskyy Ah, OK. I see now. Thanks! :)