NVIDIAGameWorks/PhysX #676: unstable Rigid body when using GPU dynamics

hadar-segal-isi, 2024-11-21:
Hi,
I tried to run the same scene with and without GPU dynamics.
The scene has one kinematic rigid sphere attached to another sphere using a distance joint.
There is no collision and no gravity and there is a fair amount of damping in the system.
When GPU dynamics is enabled the rigid body motion is significantly more chaotic.
The reason I'm interested in GPU dynamics is because I would like to have soft bodies interacting with rigid bodies (and soft bodies only work with GPU).
please see attach video and usda file

https://github.com/user-attachments/assets/4275776f-1116-4488-a611-40dbfda4bdf3
[distance-joint.usda.txt](https://github.com/user-attachments/files/17841587/distance-joint.usda.txt)

Thanks

vreutskyy, 2024-11-21:
Hi @hadar-segal-isi 
Thanks for reporting this issue.
A minor thing, but for Omniverse apps and the latest PhysX it's better to use this repository: https://github.com/NVIDIA-Omniverse/PhysX
Regards

hadar-segal-isi, 2024-11-21:
Will do, thanks!