NVIDIAGameWorks/PhysX #680: Crash in AABBTree::refitNode

Jordi-Gil, 2025-05-13:
Hello, we are getting a crash in `AABBTree::refitNode`. The callstack given is the following one:

```
PhysX_64.dll!physx::shdfnd::aos::V4LoadU(const float * const f) Line 446	C++
PhysX_64.dll!refitNode(physx::Sq::AABBTreeRuntimeNode * current, const physx::PxBounds3 * boxes, const unsigned int * indices, physx::Sq::AABBTreeRuntimeNode * const nodeBase) Line 399	C++
PhysX_64.dll!physx::Sq::AABBTree::refitMarkedNodes(const physx::PxBounds3 * boxes) Line 542	C++
PhysX_64.dll!physx::Sq::ExtendedBucketPruner::refitMarkedNodes(const physx::PxBounds3 * boxes) Line 280	C++
PhysX_64.dll!physx::Sq::AABBPruner::refitUpdatedAndRemoved() Line 822	C++
PhysX_64.dll!physx::Sq::AABBPruner::commit() Line 425	C++
PhysX_64.dll!physx::Sq::SceneQueryManager::afterSync(physx::PxSceneQueryUpdateMode::Enum updateMode) Line 415	C++
PhysX_64.dll!physx::NpScene::fetchResultsPostContactCallbacks() Line 2119	C++
PhysX_64.dll!physx::NpScene::fetchResults(bool block, unsigned int * errorState) Line 2195	C++

```

I've removed the force inline and deoptimized the code to be able to see the exact line.

In `refitNode`

`nodeBase` looks like

[image]

while `data` looks like

[image]

In line 399, we get the position based on the `nodeBase` and the `data`, but this results in invalid memory

[image]

[image]

This started happening after using the pruning structure to add all the objects to the world in a single operation

```
if (!pruningStructureActors.IsEmpty())
{
	{
		LOCK_SCENE_WRITE
		auto pruningStructure = GetPhysXSDK()->createPruningStructure(pruningStructureActors.GetPointer(), pruningStructureActors.Size());
		m_pPhysXWorld->addActors(*pruningStructure);
		pruningStructure->release();
	}
}
```

As you suggested in another thread #546 , I ran the checked version and enabled the validity function, which is not throwing anything.

Jordi-Gil, 2025-05-27:
Thanks for having a look, much appreciate it. And, no, sadly we cannot update to a newer version right now. I am still investigating if a memory stomp could be causing this problem in `PhysX`... I will update the ticket if that is the case.

> And I guess you don't have a repro either ?

No... sorry :(

Jordi-Gil, 2025-05-27:
If I add the array of actors directly without using the PruningStructure it works then.

drink18, 2026-03-13:
I'm hitting this crash recently when I tried to use PruningStructure. 
Few observations: 
- quite random crash 
- not caused by threading (I'm 99% sure )
- likely caused by some uninitialized variables somewhere (that's why it appears so randomly)