NVIDIAGameWorks/PhysX #682: BUG: physx4.1 crash on linux:  PxsNphaseImplementationContext::unregisterContactManager

OttoX, 2025-06-28:
Is this bug still not fix yet? it happens rarely so we can't find a stable way to reproduce the scenario.
Is there any possible solutions?

Here is the crash stack:

Program terminated with signal SIGSEGV, Segmentation fault.
#0  0x0000560b413158c1 in physx::PxsNphaseImplementationContext::unregisterContactManager (
    this=0x560c12529760, cm=0x0)
    at /root/Branches/l36_windows_zwne/nshm/trunk/program/implib/physx-4.1.1/physx/source/lowlevel/software/src/PxsNphaseImplementationContext.cpp:692
692     /root/Branches/l36_windows_zwne/nshm/trunk/program/implib/physx-4.1.1/physx/source/lowlevel/software/src/PxsNphaseImplementationContext.cpp: No such file or directory.
[Current thread is 1 (Thread 0x7f16354b7700 (LWP 1944206))]
(gdb) bt
#0  0x0000560b413158c1 in physx::PxsNphaseImplementationContext::unregisterContactManager (
    this=0x560c12529760, cm=0x0)
    at /root/Branches/l36_windows_zwne/nshm/trunk/program/implib/physx-4.1.1/physx/source/lowlevel/software/src/PxsNphaseImplementationContext.cpp:692
#1  0x0000560b412ae94d in physx::Sc::ShapeInteraction::destroyManager (this=0x560c14848320)
    at /root/Branches/l36_windows_zwne/nshm/trunk/program/implib/physx-4.1.1/physx/source/simulationcontroller/src/ScShapeInteraction.h:317
#2  physx::Sc::Scene::finishBroadPhaseStage2 (this=this@entry=0x560b84176020, ccdPass=1)
    at /root/Branches/l36_windows_zwne/nshm/trunk/program/implib/physx-4.1.1/physx/source/simulationcontroller/src/ScScene.cpp:6101
#3  0x0000560b412aeaf8 in physx::Sc::Scene::updateCCDSinglePassStage3 (this=0x560b84176020,
    continuation=0x560b70017880)
    at /root/Branches/l36_windows_zwne/nshm/trunk/program/implib/physx-4.1.1/physx/source/simulationcontroller/src/ScScene.cpp:3115
#4  0x0000560b411de524 in physx::Cm::Task::run (this=<optimized out>)
    at /root/Branches/l36_windows_zwne/nshm/trunk/program/implib/physx-4.1.1/physx/source/common/src/CmTask.h:66
#5  0x0000560b411669e9 in physx::Ext::DefaultCpuDispatcher::runTask (task=..., this=<optimized out>)
    at /root/Branches/l36_windows_zwne/nshm/trunk/program/implib/physx-4.1.1/physx/source/physxextensions/src/ExtDefaultCpuDispatcher.h:98
#6  physx::Ext::CpuWorkerThread::execute (this=0x560b47367308)
    at /root/Branches/l36_windows_zwne/nshm/trunk/program/implib/physx-4.1.1/physx/source/physxextensions/src/ExtCpuWorkerThread.cpp:96
#7  0x0000560b414ffb47 in physx::shdfnd::(anonymous namespace)::PxThreadStart (arg=<optimized out>)
    at /root/Branches/l36_windows_zwne/nshm/trunk/program/implib/physx-4.1.1/physx/source/foundation/src/unix/PsUnixThread.cpp:140
#8  physx::shdfnd::(anonymous namespace)::PxThreadStart (arg=0x560b46ae3a80)

vreutskyy, 2025-07-01:
Yes, seems fine.

I see the code like this in the current version
```
							//We must check to see if we have a contact manager here. There is an edge case where actors could be put to 
							//sleep after discrete simulation, prior to CCD, causing their contactManager() to be destroyed. If their bounds
							//also ceased overlapping, then this code will try to destroy the manager again.
							if(si->getContactManager())
								si->destroyManager();
```

so, perhaps it's a known issue.

OttoX, 2025-07-01:
Appreciate for your reply, thanks.

closed 2025-07-01 by OttoX

FatherTimeSDKP, 2025-07-01:
The problem if(si->getContactManager())
    si->destroyManager();