NVIDIAGameWorks/PhysX #684: Crash in nn::os::FreeTlsSlot via physx::NpPhysics::releaseSceneInternal

sabibria, 2025-08-14:
We're getting a Switch2-only assert failure with this callstack on rare occasion:

nnSdk.nss!nn::os::FreeTlsSlot(nn::os::TlsSlot)
End.nss!physx::shdfnd::TlsFree(physx::PxU32 index) Line 410 C++
End.nss!physx::NpScene::~NpScene(physx::NpScene * this) Line 186 C++
End.nss!physx::NpScene::~NpScene(physx::NpScene * this) Line 149 C++
End.nss!physx::NpPhysics::releaseSceneInternal(physx::NpPhysics * this, physx::PxScene & scene) Line 326 C++

Other platforms are fine.  Any idea what might be the cause?  Any possible workaround?  FWIW looking at the code it doesn't appear to be a double delete case.

sabibria, 2025-08-25:
I saw that this post just got closed.  I learned that this happens very reliably on Switch 2 if I ever have 13 PxScenes open at the same time, (12 of them contataining PxArticulations in my case - though not sure if that's relevant).  My workaround is just to limit the number of simultaneous scenes to 12, but please post here if you ever happen to find a code solution

ayoub-belarbi, 2025-08-27:
Hi Brian,

We’ve closed this issue since an internal ticket has already been created to track it. Additionally, as Simon mentioned, Switch 2 is not supported through this public GitHub repository. For Switch 2–related matters, please reach out via the official developer support channels instead.

Thanks,
Ayoub

sabibria, 2025-08-27:
Understood - Thank you!  I have reached out to Nintendo as well